pycom/pycom-micropython #2: canne make WiPy buils

robert-hh, 2016-11-15:
Trying to build the version for WiPy in cc3200 fails with:
`../py/obj.h:33:22: fatal error: esp_attr.h: No such file or directory
`
This file is missing in the source tree.

danicampora, 2016-11-15:
Thanks for the report. We will update the cc3200 port soon.

livius2, 2016-11-18:
In current revision ( e1a0670 few minutes ago)
i got error when i try to make WIPY
bootloader/gpio.c: In function 'gpio_set_pull_mode':
bootloader/gpio.c:181:13: error: 'PIN_PULLUP_EN' is deprecated (declared at P:/msys32/esp-idf/components/esp32/include/soc/io_mux_reg.h:56) [-Werror=deprecated-declarations]
             PIN_PULLUP_EN(GPIO_PIN_MUX_REG[gpio_num]);

danicampora, 2016-11-18:
Is that with the latest IDF from Espressif or with the one here: https://github.com/pycom/pycom-esp-idf ?

You should always user ours...

livius2, 2016-11-18:
in pycom-esp-idf/docs/windows-setup.rst
there is step 2

> Step 2: Getting the esp-idf repository from github

closed 2017-12-19 by Xykon