pycom/pycom-micropython #4: can not make esp32

YanMinge, 2016-11-15:
I got the error like this

$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/WIPY/release/bootloader/
CC bootloader/bootloader.c
CC bootloader/bootmgr.c
CC bootloader/mperror.c
CC bootloader/gpio.c
AR build/WIPY/release/bootloader/bootloader.a
LINK build/WIPY/release/bootloader/bootloader.elf
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -llog
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lspi_flash
collect2.exe: error: ld returned 1 exit status
make: *** [application.mk:270：build/WIPY/release/bootloader/bootloader.elf]

abiliojr, 2016-11-15:
We did some changes that probably fix this. Can you try it again? (warning: we had to do a push --force in the past few minutes)

YanMinge, 2016-11-15:
Yes, this issue is fixed. Thanks

closed 2016-11-15 by YanMinge