pycom/pycom-micropython #5: pull request str.rjust() str.ljust

livius2, 2016-11-15:
Hi,

i do not know how to create pull request
but will be good to add this standard functions like rjust and ljust
check it because i am not fluent in declarations like mp_obj_t... and do not know if parameter for fillchar with default value can be declared in this way
two files to change and 1 method to change 2 new to add

file: py/mpconfig.h
------------------------- after str.center
// Whether str.ljust() method provided
#ifndef MICROPY_PY_BUILTINS_STR_LJUST
#define MICROPY_PY_BUILTINS_STR_LJUST (0)
#endif

// Whether str.rjust() method provided
#ifndef MICROPY_PY_BUILTINS_STR_RJUST
#define MICROPY_PY_BUILTINS_STR_RJUST (0)
#endif

file: py/objstr.c
---------------------------------
change also existing str_center to support fillchar param
----
#if MICROPY_PY_BUILTINS_STR_CENTER
STATIC mp_obj_t str_center(mp_obj_t str_in, mp_obj_t width_in, mp_obj_t fillchar=' ') {
    GET_STR_DATA_LEN(str_in, str, str_len);
    mp_uint_t width = mp_obj_get_int(width_in);
    if (str_len >= width) {
        return str_in;
    }

    vstr_t vstr;
    vstr_init_len(&vstr, width);
    memset(vstr.buf, fillchar, width);
    int left = (width - str_len) / 2;
    memcpy(vstr.buf + left, str, str_len);
    return mp_obj_new_str_from_vstr(mp_obj_get_type(str_in), &vstr);
}
#endif

#if MICROPY_PY_BUILTINS_STR_LJUST
STATIC mp_obj_t str_ljust(mp_obj_t str_in, mp_obj_t width_in, mp_obj_t fillchar=' ') {
    GET_STR_DATA_LEN(str_in, str, str_len);
    mp_uint_t width = mp_obj_get_int(width_in);
    if (str_len >= width) {
        return str_in;
    }

    vstr_t vstr;
    vstr_init_len(&vstr, width);
    memset(vstr.buf, fillchar, width);
    int left = width - str_len;
    memcpy(vstr.buf + left, str, str_len);
    return mp_obj_new_str_from_vstr(mp_obj_get_type(str_in), &vstr);
}
#endif

#if MICROPY_PY_BUILTINS_STR_RJUST
STATIC mp_obj_t str_rjust(mp_obj_t str_in, mp_obj_t width_in, mp_obj_t fillchar=' ') {
    GET_STR_DATA_LEN(str_in, str, str_len);
    mp_uint_t width = mp_obj_get_int(width_in);
    if (str_len >= width) {
        return str_in;
    }

    vstr_t vstr;
    vstr_init_len(&vstr, width);
    memset(vstr.buf, fillchar, width);
    memcpy(vstr.buf, str, str_len);
    return mp_obj_new_str_from_vstr(mp_obj_get_type(str_in), &vstr);
}
#endif

regards,
Karol Bieniaszewski

danicampora, 2016-11-16:
@livius2 thanks for your contribution. It would be much better if done via pull-request. You only need to fork our repo, make your changes, compile and test them, and after that push to your github account. One that's done, you come to this repo and you click the **pull request** button which will guide you through the process. Let me know if you need help with that/

livius2, 2016-11-16:
first i must know what is this?
MP_DECLARE_CONST_FUN_OBJ_2(str_center_obj);
MP_DECLARE_CONST_FUN_OBJ_1(str_lower_obj);
in py/objstr.h

is it - how many parameters procedure have?
...OBJ_2 two params?
...OBJ_1 one param
and if i write 
OBJ_3 that does mean 3 params?
I am not familar with C lang and .h files
My primary language is Delphi and there are not .h files

and i do not see simple comilation steps for Windows platform

lonetech, 2016-11-16:
There exists a less efficient workaround using formatting operations. `s.ljust(n)` is equivalent to `'{:<{}}'.format(s,n)` (use `>` for right justified).

livius2, 2016-11-16:
because of that i like short writing and python standard writing
but thanks for workaround

livius2, 2016-11-18:
I see that this is too complicated to me
my C knowledge is to minimal - i see that C does not support default parameters
and above code **mp_obj_t fillchar=' '** must be changed to without =''
i do not see possibility to put this function as avaiable to board
image compailed but i got info that this is not used - and after update board i see changed version to my custom 0.9.5.**k**b1 but i can not use it

`QSTR not updated
CC ../py/objstr.c
../py/objstr.c:824:17: warning: 'str_center' defined but not used [-Wunused-function]
 STATIC mp_obj_t str_center(mp_obj_t str_in, mp_obj_t width_in, mp_obj_t fillchar) {
                 ^
../py/objstr.c:839:17: warning: 'str_ljust' defined but not used [-Wunused-function]
 STATIC mp_obj_t str_ljust(mp_obj_t str_in, mp_obj_t width_in, mp_obj_t fillchar) {
                 ^
../py/objstr.c:854:17: warning: 'str_rjust' defined but not used [-Wunused-function]
 STATIC mp_obj_t str_rjust(mp_obj_t str_in, mp_obj_t width_in, mp_obj_t fillchar) {
`

if someone can point me into direction
if not then remove this pull request

danicampora, 2016-11-18:
Please don't remove it. Thanks for your contribution. We'll turn this into a commit ourselves.

livius2, 2016-11-20:
I will look how you do this to learn self what must be done for my future pull requests

livius2, 2016-11-21:
this can be removed
i have finally finished job and created real pull request
one week study of old C and this port macros bring me to finished code
all was checked successfully on WiPy 2.0

closed 2017-12-19 by Xykon