pycom/pycom-micropython #8: LoPy: using P15 crashes

faif, 2016-11-16:
* The Pycom board you are using: LoPy
* The firmware version you are using: 0.9.4.b1
* Exact steps to cause this issue. Execute the following code:
```
import machine
machine.Pin('P15')
```
* What you expected
    * to get the Pin value
* What happened instead
    * Crash (Guru Meditation) and the CPU halted

lonetech, 2016-11-16:
Also affects P16. The problem occurs running init(), you can still access the pin objects within machine.Pin.module or machine.Pin.exp_board.

lonetech, 2016-11-16:
Erm.. I accidentally fixed this, it seems bug #10 was the root cause. That affected P15=GPIO38, which would poke at GPIO6, and P16=GPIO39, which would poke at GPIO7. I'm not sure what those do, they're not in the LoPy pin map.

commit 3dd78ef3 (2016-11-16): esp32: Fix bug when initializing and setting values of pins > 31.

The pin bit mask needs to be created explicitly using a 64-bit value,
and when setting the values of pins, we should not shift more bits
than what is actually needed.

@lonetech thanks for this discovering and solving these subtle bugs!

Fixes #8 and #10
commit fc4dd58c (2016-11-16): esp32: Fix bug when initializing and setting values of pins > 31.

The pin bit mask needs to be created explicitly using a 64-bit value,
and when setting the values of pins, we should not shift more bits
than what is actually needed.

@lonetech thanks for this discovering and solving these subtle bugs!

Fixes #8
Fixes #10
closed 2016-11-16 by danicampora