pycom/pycom-micropython #12: LoPy: wlan.ssid() always returns default network, e.g. `lopy-wlan-xxxx`

betabrain, 2016-11-17:
### Version Information:
```python
(sysname='LoPy', nodename='LoPy', release='0.9.4.b1', version='bc42ea7 on 2016-11-14', machine='LoPy with ESP32')
```
### Problem:
`wlan.ssid()` always reports the same value, no matter which network LoPy has actually joined.

```python
from network import WLAN

w = WLAN()
w.ssid() # lopy-wlan-xxxx
w.connect('homewifi', auth=(3, 'secretsecret'), timeout=5000)
w.ssid() # lopy-wlan-xxxx
w = WLAN()
w.ssid() # lopy-wlan-xxxx
w.connect('homewifi', auth=(3, 'secretsecret'), timeout=5000)
w.ssid() # lopy-wlan-xxxx
w = WLAN()
w.ssid() # lopy-wlan-xxxx
```

betabrain, 2016-11-17:
I wonder if it would be preferrable to have a method `wlan.current()` (or something like it) which returns a named tuple (in lines with what is returned by `wlan.scan()`). Currently there are two different conventions at play, named tuples, and getters, which is confusing.

```python
mynet = wlan.current()
type(mynet) # tuple
mynet.ssid # no ()
```

Alternatively, the getters could be replaced with Python properties to make them look and feel like the read-only vars on those named tuples.

```python
wlan.scan()[0].ssid # no ()
wlan.ssid # also no ()
```

These are just suggestions.

danicampora, 2016-11-18:
Thanks for the report. We are working on the fix. Should be available on the release from today a bit later.

> Alternatively, the getters could be replaced with Python properties to make them look and feel like the read-only vars on those named tuples.

Properties would be nice, but we opted not to use them in MicroPython for efficiency reasons. We had a long discussion on the MicroPython repo about that like a year ago...

danicampora, 2016-11-18:
Fixed with https://github.com/pycom/pycom-micropython/commit/0c3cee9c27b6878554196e9898540ca0d2a59f47

closed 2016-11-18 by danicampora

betabrain, 2016-11-18:
Great! Just realized that `wlan.auth()` seems to have a related problem. I'm currently connected to my home network, but `wlan.auth()` returns the this:

```python
>>> wlan.auth()
(3, 'www.pycom.io')
```

betabrain, 2016-11-20:
`wlan.auth()` issue is still present in:
```python
(sysname='LoPy', nodename='LoPy', release='0.9.5.b1', version='v1.8.6-13-ge1a0670 on 2016-11-18', machine='LoPy with ESP32')
```

Please reopen.