pycom/pycom-micropython #14: LoPy Firmware 0.9.5.b1 returns OSError:11

kruisdraad, 2016-11-19:
AFter upgrading the LoPy to 0.9.5.b1 and using the Lora ABP code from the docs (p.s. import binascii and import struct are missing there?) i am getting the OSError:11 at the moment i am trying to send a lora packet (s.send('bla'))

Although the blue led keeps blinking the PyMakr console dies and no commands can be given anymore. Restarting the PyMakr will just report it cannot communicate with the device. Resetting the device and all things are fine again. I can run code a few (random number, but not more then 5) times before this error returns. With run i am your ABP example with 1x send and click on run, wait for console to return and run again.

Perhaps some kind if 'leak' running the code multiple times (socket not cleared properly?) ... running the send in a loop works fine.

Is there a table explaining alle the OSErrors?

kruisdraad, 2016-11-19:
# make the socket non-blocking
s.setblocking(False)

while True:
    print("sending")
    s.send('hello')
    print("done")
    sleep(2)

returns in a OSError 11 too after 7 iterations. Using a blocking method then it works fine

danicampora, 2016-11-20:
OSError 11 means **EAGAIN**, which happens because you are trying to send too fast. Blocking sockets do not show the error because the call doesn't return until the LoRa packet is completely sent. Sending a LoRa packet takes several seconds.

You need to catch this exception, and simply wait before trying to send again, for instance:

```python
import time

s.setblocking(False)

while True:
    try:
        print("sending")
        s.send('hello')
        print("done")
    except Exception as e:
        if e.args[0] == 11:
            print('cannot send just yet, waiting...')
            time.sleep(0.5)
        else:
            raise    # raise the exception again
    time.sleep(2.0)
```

I know that this is missing from the docs, we are working on it...

On the next release we will add the **ernoo** module, and add this feature to MicroPython (which was rejected by one of the maintainers, but deserves to be added):

https://github.com/micropython/micropython/pull/2407/commits

which will allow you to do:

```python
import errno

...
...

except Exception as e:
    if e.errno == errno.EAGAIN:
        print('cannot send just yet, waiting...')
        time.sleep(0.5)
```

Which is a lot nicer and readable :-)

kruisdraad, 2016-11-20:
Well ill recheck this, but so far i can reproduce this without sending. The first attempt i just did by stopping the code and then waiting for 10 minutes and it still refuses to send anything until a full reset is done.

Every time i run code, CTRL-C it ... change the code and work for like 2-3 minutes, and then start the code up again returns in a OSError11. I would assume that even waiting more then 5 seconds would free up the radio in normal conditions.  Doing this multiple times (20+) makes the unit fully unresponsive.

I added the catching code, which results in a endless 'cannot send'. Setting it to non-blocking does give moments it says i can send , but its not sending anything. No gateways received any data (not even non-CRC) and i pulled out a SDR and no signals detected from the unit at all. Again resetting it and running the code for the first time works fine.

danicampora, 2016-11-21:
@kruisdraad if that's the case then something is definitely broken. I'll test and report ASAP. Thanks.

danicampora, 2016-11-21:
@kruisdraad I'm trying to reproduce your issue and when joining using OTAA I can't. Did you join using ABP?

danicampora, 2016-11-21:
> @kruisdraad I'm trying to reproduce your issue and when joining using OTAA I can't. Did you join using ABP?

Oh yes you did, I just re-read your original post. Will test now...

kruisdraad, 2016-11-21:
I can send the original code used and show you how to reproduce it, but it is ABP indeed

danicampora, 2016-11-21:
@kruisdraad please do send me all your original code, as I cannot reproduce it with ABP either, and I can see all my packets on the Gateway.

danicampora, 2016-11-21:
I do get the OSError 11 from time to time, but catching the exception as I suggested works without problems.

danicampora, 2016-11-21:
@kruisdraad are you using our binary from the update tool, or have you built the software yourself?

kruisdraad, 2016-11-21:
what do you mean with firmware? I downloaded the latest (windows) PyMakr and update tool and run those

I turn on the device. Then i open PyMakr, build some code and press start/run (so not uploading it to the storage, main.py) runs fine, no OSErrors. Then i CTRL-C to stop the running code (break the loop) change stuff and press run again. Then no packets at all with blocking and nonblocking just errors. Resets works fine.

I think the difference might be the code is still in memory as i can so s.send('') on the console and running the code again did not reset (init) the lora radio again or something similair. Then again doing a s.send on the console returns a OSError in 40% of the time, but 100% no actual packets send.

kruisdraad, 2016-11-21:
Partial code with Wifi and other stuff stripped, which i used to test this yesterday to make sure it wasnt other code.

```
from network import LoRa
from time import sleep

import socket
import binascii
import struct

# Initialize LoRa in LORAWAN mode.
lora = LoRa(mode=LoRa.LORAWAN, sf=7, tx_power=14)

lora.BW_125KHZ
lora.CODING_4_5

# create an ABP authentication params
dev_addr = struct.unpack(">l", binascii.unhexlify('xx'.replace(' ','')))[0]
nwk_swkey = binascii.unhexlify('xx'.replace(' ',''))
app_swkey = binascii.unhexlify('xx'.replace(' ',''))

# join a network using ABP (Activation By Personalization)
lora.join(activation=LoRa.ABP, auth=(dev_addr, nwk_swkey, app_swkey))

# create a LoRa socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# set the LoRaWAN data rate
s.setsockopt(socket.SOL_LORA, socket.SO_DR, 5)

# make the socket non-blocking
s.setblocking(True )

while True:
    try:
        print("sending")
        s.send('hello')
        print("done")
    except Exception as e:
        if e.args[0] == 11:
            print('cannot send just yet, waiting...')
            sleep(2)
        else:
            raise 

```

danicampora, 2016-11-21:
> what do you mean with firmware? I downloaded the latest (windows) PyMakr and update tool and run those

OK, that's good. I thought that maybe you were compiling from source.

> I turn on the device. Then i open PyMakr, build some code and press start/run (so not uploading it to the storage, main.py) runs fine, no OSErrors. Then i CTRL-C to stop the running code (break the loop) change stuff and press run again. 

That might be it... I think there's an issue at the moment when "trying" to initialize the LoRa radio twice.

danicampora, 2016-11-21:
> I turn on the device. Then i open PyMakr, build some code and press start/run (so not uploading it to the storage, main.py) runs fine, no OSErrors. Then i CTRL-C to stop the running code (break the loop) change stuff and press run again.

Yes, confirmed, I can reproduce it, the issue is when initializing the LoRa radio again before a hard reset. We'll fix this.
@kruisdraad thanks for the report!

Cheers,
Daniel

kruisdraad, 2016-11-21:
Thnx, i found 3 more issues ... once i am back home again and can confirm/reproduce them ill document them and send some more tickets along.

danicampora, 2016-11-21:
@kruisdraad perfect, please do so, thanks!