pycom/pycom-micropython #17: MemoryError when starting a new thread

betabrain, 2016-11-20:
### Version Info
```python
(sysname='LoPy', nodename='LoPy', release='0.9.5.b1', version='v1.8.6-13-ge1a0670 on 2016-11-18', machine='LoPy with ESP32')
```

### Issue
```python
>>> import time
>>> import _thread
>>> import gc
>>> def hello(n):
...     time.sleep(2)
...     print('hello', n)
...
>>> gc.mem_free()
38384
>>> _thread.start_new_thread(hello, ('there',))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 8192 bytes
```

danicampora, 2016-11-24:
I cannot reproduce it just like that. What else did you do to have only **38384** bytes free of memory?

betabrain, 2016-11-25:
I have built an init daemon of sorts that scans two directories and runs setup scripts as well as run asyncio "services". I then exit the main loop with `^C` and run above code. Ignoring that, there shouldn't be a MemoryError with 38K free.

I managed to change my scripts to fail to reproduce this issue, but so far I don't have a minimal example. I will get back to you, either with the actual source I am running, or a minimal example.

betabrain, 2016-11-26:
The following seems to work each time for me:
#### lib/sha256.py
The pure-python implementation from [micropython/micropython-lib's hashlib](https://github.com/micropython/micropython-lib/blob/6c0c7f84cb320dc580179291f00c1ec2b02f8097/hashlib/hashlib/sha256.py). Get the raw file [here](https://github.com/micropython/micropython-lib/raw/6c0c7f84cb320dc580179291f00c1ec2b02f8097/hashlib/hashlib/sha256.py) for convenience.

#### main.py
```python
import gc
print('after import gc', gc.mem_free())
import sha256
print('after import sha256', gc.mem_free())
import time
import _thread
print('after remaining imports', gc.mem_free())

def hello(n):
    time.sleep(2)
    print('hello', n, gc.mem_free())
print('after def hello()', gc.mem_free())

print('before 1', gc.mem_free())
_thread.start_new_thread(hello, ('there-1',))

print('before 2', gc.mem_free())
_thread.start_new_thread(hello, ('there-2',))

print('before 3', gc.mem_free())
_thread.start_new_thread(hello, ('there-3',))

print('before 4', gc.mem_free())
_thread.start_new_thread(hello, ('there-4',))

print('before 5', gc.mem_free())
_thread.start_new_thread(hello, ('there-5',))

print('before 6', gc.mem_free())
_thread.start_new_thread(hello, ('there-6',))
```

#### Failing Output
```python
after import gc 71008
after import sha256 59552
after remaining imports 59552
after def hello() 59536
before 1 59536
before 2 50896
before 3 42256
Traceback (most recent call last):
  File "main.py", line 21, in <module>
MemoryError: memory allocation failed, allocating 8192 bytes
MicroPython v1.8.6-13-ge1a0670 on 2016-11-18; LoPy with ESP32
Type "help()" for more information.
>>> hello there-1 44464
hello there-2 44464

>>>
```
There should be plenty of memory (`42256`) to start thread 3, but it fails. Strangely, trying to start a thread later, by hand, does work! 😕 

```python
>>> dir()
['time', 'gc', '__name__', '_thread', 'sha256', 'hello']
>>> gc.mem_free()
44224
>>> _thread.start_new_thread(hello, ('there-7',))
>>> hello there-7 37600

>>>
```

#### Successful Output
Commenting out `import sha256` on line 3 disables the error:
```python
after import gc 71024
after import sha256 71024
after remaining imports 71024
after def hello() 71008
before 1 71008
before 2 62368
before 3 53728
before 4 45088
before 5 36448
before 6 27808
MicroPython v1.8.6-13-ge1a0670 on 2016-11-18; LoPy with ESP32
Type "help()" for more information.
>>> hello there-2 19136
hello there-3 19136
hello there-1 19136
hello there-5 19136
hello there-6 19136
hello there-4 19136

>>>
```

@danicampora Can you reproduce this?