pycom/pycom-micropython #18: Building LoPy fails, instructions might be wrong

ejpir, 2016-11-20:
Something might be wrong with the build instructions. I get an -lcore ld error, both on ubuntu and macOS Sierra. Could someone please look into this? Thanks!

# $IDF_PATH
nick:esp32 nick$ echo $IDF_PATH
/Users/nick/LoPy/pycom-esp-idf

nick:esp32 nick$ ls /Users/nick/LoPy/pycom-esp-idf
CONTRIBUTING.rst  Kconfig  LICENSE  README.md  add_path.sh  components	docs  examples	make  tools

# $PATH
nick:esp32 nick$ echo $PATH
/usr/local/opt/coreutils/libexec/gnubin:/usr/local/opt/gnu-sed/libexec/gnubin:/Users/nick/google-cloud-sdk/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/go/bin:/usr/local/MacGPG2/bin:/Library/Java/JavaVirtualMachines/jdk1.8.0_60.jdk/Contents/Home/bin:/Applications/Atlassian/atlassian-plugin-sdk-5.1.1/apache-maven-3.2.1/bin:/Users/nick/LoPy/xtensa-esp32-elf/bin

nick:esp32 nick$ ls /Users/nick/LoPy/xtensa-esp32-elf/bin

xtensa-esp32-elf-addr2line  xtensa-esp32-elf-ct-ng.config  xtensa-esp32-elf-gcc-ranlib	xtensa-esp32-elf-objcopy
xtensa-esp32-elf-ar	    xtensa-esp32-elf-elfedit	   xtensa-esp32-elf-gcov	xtensa-esp32-elf-objdump
xtensa-esp32-elf-as	    xtensa-esp32-elf-g++	   xtensa-esp32-elf-gdb		xtensa-esp32-elf-ranlib
xtensa-esp32-elf-c++	    xtensa-esp32-elf-gcc	   xtensa-esp32-elf-gprof	xtensa-esp32-elf-readelf
xtensa-esp32-elf-c++filt    xtensa-esp32-elf-gcc-4.8.5	   xtensa-esp32-elf-ld		xtensa-esp32-elf-size
xtensa-esp32-elf-cc	    xtensa-esp32-elf-gcc-ar	   xtensa-esp32-elf-ld.bfd	xtensa-esp32-elf-strings
xtensa-esp32-elf-cpp	    xtensa-esp32-elf-gcc-nm	   xtensa-esp32-elf-nm		xtensa-esp32-elf-strip

# make fails with -lcore

nick:esp32 nick$ make BOARD=LOPY LORA_BAND=USE_BAND_868
LINK build/LOPY/release/bootloader/bootloader.elf
xtensa-esp32-elf-gcc -nostdlib -Wl,-Map=build/LOPY/release/bootloader/bootloader.map -Wl,--no-check-sections -u call_user_start_cpu0 -Wl,-static -Wl,--undefined=uxTopUsedPriority -Wl,--gc-sections -T esp32.bootloader.ld -T esp32.rom.ld -T esp32.peripherals.ld -Wl,--start-group -Lbootloader/lib -L/Users/nick/LoPy/pycom-esp-idf/components/esp32/ld -Lbuild/LOPY/release/bootloader -L/Users/nick/LoPy/pycom-esp-idf/components/esp32/lib -Lbootloader -llog -lcore -lspi_flash build/LOPY/release/bootloader/bootloader.a -Wl,--end-group -Wl,-EL -o build/LOPY/release/bootloader/bootloader.elf
/Users/nick/LoPy/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/ld: cannot find -lcore
collect2: error: ld returned 1 exit status
make: *** [build/LOPY/release/bootloader/bootloader.elf] Error 1

danicampora, 2016-11-20:
You also need to checkout all the sub-modules within the IDF. Those libcore.a library and several other are part of a submodule. Please see the details of the latest commit to the README:

https://github.com/pycom/pycom-micropython/commit/7c37117dc29d2a4e9764e713eeab6bcb4c61fa8a

closed 2016-11-20 by danicampora

ejpir, 2016-11-20:
@danicampora thanks this works. it was kind of unclear for me as a newcomer, because those dependencies are listed as if they are for the Unix port.

Might want to update the docs to add that section under the ESP32 port as well?

Either way, thanks for the quick reply.