pycom/pycom-micropython #21: newest firmware is not booted after flashing another custom firmware

ejpir, 2016-11-21:
I have successfully flashed a custom built firmware that was built from this git. Booted without problems.

After removing the changes and reflashing a new firmware, the old firmware changes are still there and it seems the 2nd custom firmware is never loaded.

Tried to fix this by:
- change the makeversionhdr.py script to generate custom versions with "-nick" appended. No luck there.
- flash using serial. no luck

```
Something seems wrong. I have even updated the py
/ This file was generated by py/makeversionhdr.py
#define MICROPY_GIT_TAG "v1.8.6-16-g7c37117-dirty-nick"
#define MICROPY_GIT_HASH "7c37117-dirty-nick"
#define MICROPY_BUILD_DATE "2016-11-21"
#define MICROPY_VERSION_MAJOR (1)
#define MICROPY_VERSION_MINOR (8)
#define MICROPY_VERSION_MICRO (6)
#define MICROPY_VERSION_STRING "1.8.6-nick"
```

How does the Lopy determine what the newest firmware is?

ejpir, 2016-11-21:
@danicampora I got it to load the newest firmware by pulling P12 up and releasing it in the first 3 second window.

**Is this a bug? I thought it should boot the newest in normal mode as well.**

Now it is booting the newest firmware in normal mode, even after totally powering down.

Before doing the P12 thing it was not loading the latest firmware.

danicampora, 2016-11-24:
Combining OTA updates with serial flasher updates gives estrange results. These two methods of flashing are not meant to be used together. That's why our serial flasher erases almost the entire flash (except for the file system and the LoRa MAC address), an the flashes the bootloader, the partition table.

ejpir, 2016-11-24:
oke. I have had more luck since then when only using the serial flasher :)

closed 2017-01-31 by abiliojr