pycom/pycom-micropython #24: Duty cycle calculations seem to be off

kruisdraad, 2016-11-21:
When transmitting a single packet with 1 byte payload (+overhead) your not able to send as much packets like any other unit with LoRa. I am not sure if its related to the very large time to send a packet (3s) or something else that offsets the calculation.

This is not a big issue, but with a total time between packets at best 3.87s vs highest other at 2,04s its nearly double the sample interface in case you want to map an area.

kruisdraad, 2016-11-21:
In addition when using high payloads the duty cycle should adhere that i cannot send for another few minutes (max payload size) but it still allows me to send every 3.87 seconds which allows me to go far beyond the allowed duty cycle

ejpir, 2016-11-21:
@kruisdraad I think the problem with the duty cycles is in esp32/mods/modlora.c.

```
 453                    #if defined( USE_BAND_868 )
 454                        LoRaMacTestSetDutyCycleOn(false);   // Test only
 455                        LoRaMacChannelAdd(3, (ChannelParams_t)LC4);
 456                        LoRaMacChannelAdd(4, (ChannelParams_t)LC5);
 457                        LoRaMacChannelAdd(5, (ChannelParams_t)LC6);
 458                        LoRaMacChannelAdd(6, (ChannelParams_t)LC7);
 459                        LoRaMacChannelAdd(7, (ChannelParams_t)LC8);
 460                        LoRaMacChannelAdd(8, (ChannelParams_t)LC9);
 461                        LoRaMacChannelAdd(9, (ChannelParams_t)LC10);
 462                    #endif
```

Not sure if that is in the distributed firmwares, but looks like they forgot to set it to True after testing.