pycom/pycom-micropython #25: Not able to select SF12/DR0 anymore

kruisdraad, 2016-11-21:
Before a few firmwares ago you were stuck with SF12 and not able to select anything else. Now you have options like:

`lora = LoRa(mode=LoRa.LORAWAN, sf=12, tx_power=14)`

and

`s.setsockopt(socket.SOL_LORA, socket.SO_DR, 0)`

however using them either results in these errors:

OSError: 114
OSError: the requested operation failed (< this one is funny, since Daniel explained it could only do numbers)

Or again saying that its send (no errors) but not actually sending anything. The documentation is not reflect the ability to use these functions so they may simply not be implemented, but if they are i was not able to change the datarate/SF on whatever method i tried.

danicampora, 2016-11-22:
Related to this: https://github.com/pycom/pycom-micropython/issues/26

danicampora, 2016-11-24:
Fixed with this: https://github.com/pycom/pycom-micropython/commit/d1dd08b41466569a4928d7b0e2271e3aefbfdee5

closed 2016-11-24 by danicampora