pycom/pycom-micropython #26: Setting socket.So_DR to non-existing value results in error 109

kruisdraad, 2016-11-21:
When changing

lora = LoRa(mode=LoRa.LORAWAN, sf=12, tx_power=14)

to

lora = LoRa(mode=LoRa.LORAWAN, sf=1213, tx_power=14)

you get a correct error that the value is out of range. However this does not happen with:

s.setsockopt(socket.SOL_LORA, socket.SO_DR, 99)

which results in

OSError: 109

danicampora, 2016-11-22:
Thanks, I also found this myself when testing and already have a fix for it. The fix will be available in the next release.

danicampora, 2016-11-24:
Actually, the OSError is OK since it's a socket operation. It will be changed to: 
OSError: [Errno 95] EOPNOTSUPP

Now that we have corrected the errno and added the errno module.

danicampora, 2016-11-24:
Resolved with:
https://github.com/pycom/pycom-micropython/commit/2eb2e02dcc86ef787bd17712ae9eed1a17f21c82

and:

https://github.com/pycom/pycom-micropython/commit/32aa6bf6332cc202d8eaeaf8114cbe115f6fac86

closed 2016-11-24 by danicampora