pycom/pycom-micropython #35: xQueueGenericReceive  -  assert failed!

ejpir, 2016-11-24:
Using a LoPy, custom firmware (nothing but printf statements added) built from Github.

Error occurs while sending a LoraWan message as shown below.

# version
```
os.uname().release
'0.9.5.b1'
```
# command
```
import socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)
s.setsockopt(socket.SOL_LORA, socket.SO_DR, 5)
s.setblocking(True)
s.send(bytes([1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,11,1,1,1,1,1,1,1,1,1,11,1,1,1,1,1,1,1,1,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3]))
```

# error
```
/home/danicampora/Code/pycom/esp-idf-nov-3/components/freertos/./queue.c:1443 (xQueueGenericReceive)- assert failed!
Guru Meditation Error: Core   0 panic'ed (Unhandled debug exception)
Register dump:
PC      :  40009203  PS      :  00060a36  A0      :  80084c38  A1      :  3ffd5280  
A2      :  00000075  A3      :  00000075  A4      :  00000010  A5      :  ffffffff  
A6      :  00000000  A7      :  ffffffeb  A8      :  00000000  A9      :  3ffd51f0  
A10     :  00000000  A11     :  3f400974  A12     :  3ffd51bf  A13     :  00000033  
A14     :  00000000  A15     :  3ffd51c5  SAR     :  00000004  EXCCAUSE:  00000001  
EXCVADDR:  00000000  LBEG    :  400014fd  LEND    :  4000150d  LCOUNT  :  fffffff8  
CPU halted.
```

danicampora, 2016-11-24:
Thanks. Can you share the github tag?

ejpir, 2016-11-24:
MicroPython v1.8.6-15-g50c448d-dirty