pycom/pycom-micropython #40: cc3200: AYT not supported

robert-hh, 2016-11-26:
The AYT protocol is not supported by this version. Then, pymakr cannot connect to WiPy 1 by WiFi. The AYT protocol is supported in version 1.2, but somehow the fixes of the version were never transferred to the more recent versions.

niallp, 2017-01-29:
I'm getting lost where (if anywhere) development on the WiPy 1 is happening. I just came on this repo by pycom while previously was following wipy/wipy (which is at 1.3.0 using micropython 1.8.2) and also tried micropython/micropython at latest (1.8.7) building for the WiPy (cc3200) (builds but ftp server not working).

This fork seems to follow the micropython mainline (version numbering and hashes match) though not fully up to date, but still the latest WiPy 1 firmware (1.3.0) is in wipy/wipy

Can any of the maintainers comment ? @pfalcon , @danicampora 

thx.

robert-hh, 2017-01-29:
Hello @nialip,

I just gave it a try with the build from micropython/micropython. No problem. I can build & run the WiPY 1 version, and connect to it with ftp & telnet. You have to use passive mode of  ftp, which is on Windows only supported in tools like FileZilla and FireFtp. Please take i account that in the default set-up WiPy1 starts as access point. So you have to attach to it using the proper address.
The repositories on Pycom (aka WiPy) and MicroPython differ in details, but ftp should be fine with either version.
And yes, since the ports for the esp chips are available, the cc3200 ports loses support.
Kind Regards, Robert

niallp, 2017-01-29:
@robert-hh ... thanks for giving it a try, I'll have to dig deeper on the ftp issue ... as I had it running OK with telnet as an endpoint on my LAN with my logger script I didn't try resetting to default wifi.

The release notes on wipy/wipy indicate v1.3 should support AYT but likely it wasn't merged back into micropython/micropython, I'll see if I can figure it out from respective code bases.

robert-hh, 2017-01-30:
Hello @nialip, I looked for that too, but it seemed that the source on wipy/wipy and the binaries do not match. I did not find any indication of AYT support in the sources, different form what you can see for instance in the eps32 branch. Maybe you are more successful. 
And there was a discussion about merging code from pycom into micropython, since they use a different open source policy. Micropython uses MIT, while pycom switched to GNU GPL, and thus micropython refuses to merge code under GNU GPL.

niallp, 2017-01-30:
On 2017-01-29 10:22 PM, robert-hh wrote:
>
> Hello @nialip, I looked for that too, but it seemed that the source on 
> wipy/wipy and the binaries do not match. I did not find any indication 
> of AYT support in the sources, different form what you can see for 
> instance in the eps32 branch. Maybe you are more successful.
>
Unfortunately not, I can't build anything from the wipy/wipy or other 
repos that has a working ftp server so each time I need to revert to the 
factory image to reload. I did try with the board as AP instead of 
endpoint, no difference to the ftp behaviour ... it connected the first 
time, then hangs after I set the user as 'micro' whereas the binaries 
from wipy/wipy seem to be OK.

It seems that the submodule was never updated on the 1.3.0 or 1.2.0 
releases and the version.h file not updated beyond 1.2.0 in the 
micropython source. I'm not sure where the version string is generated 
from yet, local builds don't match the <micropython-tag>-<build>-<hash> 
format that the binaries have.

In the end I have a board that I can't reproduce the firmware for that 
shows long term stability issues (my logger code dies randomly) and I'm 
not sure where to start with the source ... wipy/wipy has the most 
functional binaries but not all the source is there.

@robert-hh ... you mentioned that you had no issues with a 
micropython/micropython build, could you let me know which version ? My 
ftp server is definitely not working properly, if the source is 
identical then I may have a compiler issue (building with default 
toolchain on Ubuntu 16.04)
>
> And there was a discussion about merging code from pycom into 
> micropython, since they use a different open source policy. 
> Micropython uses MIT, while pycom switched to GNU GPL, and thus 
> micropython refuses to merge code under GNU GPL.
>
Oh well, that may well explain the forking and confusion, thanks.

             ... Niall

robert-hh, 2017-01-30:
Hello @niallp. Sad to hear about your struggle. I just had built the version from that repository: https://github.com/micropython/micropython, which is version  1.8.7-1xx, and the have pre-built images here: http://micropython.org/download.
If ftp is not running, I do not know how to upload the firmware. Never did that.
What kind of ftp client are you using? Did you try plain vanilla command line ftp on linux or OS X? The Windows command line ftp is pretty incompatible, and supports active mode only. There, my best experience is with the FireFTP add-on of Firefox. But Filezilla should be OK too, although a little tricky to configure. In any case, you have to set the connection to a single session, passive mode, and you must not use the quick mode of Filezille. You have to define a server first in file->server_manager with the right properties (1 lsession, passive mode), and then connect to it.

Robert

dpgeorge, 2017-01-30:
Daily builds of micropython/micropython repo for wipy are available here: https://micropython.org/download/#wipy .  FTP works for me on those builds (but you need to use passive mode).

You can also flash the firmware via the serial (UART) port using the Linux cc3200tool.  It works pretty well, quicker and more reliable than over FTP.  See the micropython/micropython:cc3200/README.md for details.  You will need to solder a small wire on the back of the wipy, on the SOP2 pin, and connect it to 3V3, to get it to enter the serial bootloader.

niallp, 2017-01-31:
@robert-hh  ... I don't think the issue is my ftp client, just using the regular CLI ftp from the Linux netkit (based of BSD utilities I think). It works fine (passive mode of course) on any binary firmware release from wipy/wipy and as it turns out, also with the binaries from micropython/micropython (thx @dpgeorge)

Building it locally generates no errors and the image seems to work fine with telnet and my MQTT logger script but the ftp server fails consistently after inputting user 'micro' ... something about my build process or toolchain is not right as I am using exactly the same source tree as the daily builds for my tests.

I'm running Ubuntu 16.04 x64, with the default arm-none-eabi- toolchain from gcc-arm-embedded ver. 6-2016q4-1~x ... any other ideas out there ? ...

robert-hh, 2017-01-31:
Did you try to connect with ftp to other servers, just to sort out your client & network as the cause of your trouble?

niallp, 2017-01-31:
Yes, the ftp client works with other servers and also works fine with 
the distributed binaries for the WiPy (whether from wipy/wipy or 
micropython/micropython). It seems the pycom/pycom-micropython fork 
doesn't build at all for the cc3200 (some esp32 dependency crept in) and 
I haven't found binaries from it for the WiPy 1 (their documentation 
points back to the wipy/wipy repo for support).

At this point it seems micropython/micropython offers the best support 
for the wipy 1, just need to figure out why I have problems with my 
local builds. What development environment are you using ? Maybe I 
should create a virtual dev env with whatever works ...

robert-hh, 2017-01-31:
I do not work with the pycom branch for CC3200 on github wipy/wipy. It's outdated and inconsistent. The miciropython branch is more recent, with the exception of AYT support. But that is required for pymakr only, and I'm not using that.
The system I'm using is debian Jessie, and the development environment is set-up straight along the documentation at the MicroPython web site and the CC3200 section.

niallp, 2017-02-10:
@robert-hh ... I'll move this thread to the micropython/micropython repo as it is more applicable there. Still getting the same trouble with local builds even in a clean Jessie install.

robert-hh, 2017-02-11:
The question is more: is anyone seriously maintaining WiPy 1, or did it get an obsolete product. The AYT issue is only of concern if you use Pymakr. Since users of the the micropython/micropython rarely use Pymakr, and Pycom.io users haver their focus on the new products, the whole issue seems obsolete, meaning, you can as well close it.

niallp, 2017-02-12:
Agreed, AYT support for the WiPy1 is likely a will not fix ... I don't think I have the ability to close as a mere commentor, perhaps you can as the originator ?

robert-hh, 2017-02-12:
I do not expect any progress, so let's keep the list small.

closed 2017-02-12 by robert-hh