pycom/pycom-micropython #42: Build from clean source passes, but causes LoPy to crash

brotherdust, 2016-11-26:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy): LoPy
* The firmware version you are using. You can get it by executing the following Python code at the REPL: REPL is not possible at this point. Board fails to enter REPL. I have successfully compiled and used code from previous commits without issue. Something changed in recent commits.

* Exact steps to cause this issue
1. Cloned latest pycom-micropython (9633e598dd52224d01d7728bd61c2719306e2c72) and pycom-provided espressif IDF repos (918ae7aa10852c78a3d6c7e6a7f6d7df30d2449b)
1. Compiled firmware from clean (unmodified source)
```
make BOARD=LOPY LORA_BAND=USE_BAND_915
```
2. Flashed firmware into LoPy
3. LoPy crashes

* What you expected
    * LoPy should enter REPL
* What happened instead
    * Crashy reboot loop!
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff9010,len:8
load:0x3fff9018,len:312
load:0x40078000,len:2348
ho 0 tail 12 room 4
load:0x4009f000,len:1364
entry 0x4009f2f8
˜øXTAL 40Mct 20 2016 12:36:18
I (1088) phy: phy_version: 246, Nov 18 2016, 17:30:07, 0, 2

tcpip_task_hdlxxx : 3ffdf694, prio:18,stack:2048
W (1096) wifi: wifi nvs_open fail ret=4353

/home/danicampora/Code/pycom/esp-idf-nov-22/components/freertos/./queue.c:716 (xQueueGenericSend)- assert failed!
Guru Meditation Error: Core   0 panic'ed (Unhandled debug exception)
Register dump:
PC      :  40008155  PS      :  00060d36  A0      :  80007d16  A1      :  3ffddfe0
A2      :  00000072  A3      :  00000072  A4      :  00000010  A5      :  ffffffff
A6      :  00000000  A7      :  ffffffee  A8      :  00000000  A9      :  3ffddf50
A10     :  00000000  A11     :  3f400b01  A12     :  3ffddf1f  A13     :  00000036
A14     :  00000000  A15     :  3ffddf24  SAR     :  00000004  EXCCAUSE:  00000001
EXCVADDR:  00000000  LBEG    :  4000c349  LEND    :  4000c36b  LCOUNT  :  ffffffff
```
Firmware attached.
[appimg.zip](https://github.com/pycom/pycom-micropython/files/614472/appimg.zip)

Safe-mode boot *does not work*!

Thank you!

danicampora, 2016-11-26:
We have made some changed to the bootloader and partition table that require you to erase first update with our software update tool. Get the updater here: https://www.pycom.io/support/supportdownloads/ (Under Firmware Updates).

brotherdust, 2016-11-26:
Ok. I will try it now.

brotherdust, 2016-11-26:
Success! Thanks for the info!

closed 2016-11-26 by brotherdust