pycom/pycom-micropython #43: UNIX: build fail

david-guenault, 2016-11-28:
* WiPy2
* (sysname='WiPy', nodename='WiPy', release='0.9.6.b1', version='v1.8.6-40-gd10463e on 2016-11-25', machine='WiPy with ESP32'

I'm currently trying to build pycom version of micropython on **unix**. Following README file:

```
$ cd unix
$ make axtls
$ make
```

The build fail (make) with lots of messages related to missing header esp_attr.h. Here is an example 

```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/qstr.i.last
In file included from ../py/mpstate.h:35:0,
                 from ../py/runtime.h:29,
                 from ../extmod/modbtree.c:33:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_close.c:48:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_conv.c:45:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_debug.c:47:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_delete.c:47:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_get.c:47:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_open.c:58:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_overflow.c:47:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_page.c:42:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_put.c:48:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_search.c:45:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_seq.c:48:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
compilation terminated.
In file included from ../lib/berkeley-db-1.xx/PORT/include/filevtable.h:38:0,
                 from ../lib/berkeley-db-1.xx/PORT/include/db.h:44,
                 from ../lib/berkeley-db-1.xx/btree/bt_split.c:48:
../py/obj.h:33:22: fatal error: esp_attr.h: Aucun fichier ou dossier de ce type
```

Notes:
- disabling ffi still produce the same error.

faif, 2016-11-30:
Hi @david-guenault,

Can you please be more specific about the operating system version/distribution that you are using?

Thanks

david-guenault, 2016-11-30:
OS: Ubuntu 16.04
Kernel: Linux david-laptop 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

Thx

Xykon, 2016-12-20:
You are trying to build micropython for unix instead of esp32.

You should use:

cd esp32
make BOARD=WIPY

See README.md for more details to setup the toolchain and esp-idf.

niallp, 2017-02-10:
@Xykon ... I don't think it is unreasonable to have expected this fork of micropython to still build for platforms other than esp32 (at least I made the same error ;), perhaps the README should be updated to reflect this repo *only* supports esp32 and for all other platforms go to micropython/micropython.

Xykon, 2017-12-19:
Closing this issue as we have removed references to the unix version from the readme.

closed 2017-12-19 by Xykon

nevercast, 2017-12-20:
@Xykon Does this mean the Unix version is no longer supported by Pycom? (Or never was)