pycom/pycom-micropython #46: Interrupts cause System Crash

jubbs, 2016-11-30:
(sysname='LoPy', nodename='LoPy', release='0.9.6.b1', version='v1.8.6-40-gd10463e on 2016-11-25', machine='LoPy with ESP32')

I played with interrupts when I first got the LoPy and they worked fine but I have come back to it and I am getting kernel dumps. I have tried different pins but I am sure that G17 button on the expansion board worked for me before.

Firmware 9.6.b1.

Steps to recreate

>>> import micropython
>>> from machine import Pin
>>> micropython.alloc_emergency_exception_buf(100)
>>> int_counter =0
>>> def int_handler(pin_obj):
...     global int_counter
...     int_counter+=1
...     
...     
... 
>>> button=Pin('G17', mode=Pin.IN, pull=Pin.PULL_UP)
>>> button.irq(trigger=Pin.IRQ_RISING|Pin.IRQ_FALLING, handler=int_handler)
<irq>
-----> Push the button on the expansion board.

>>> Guru Meditation Error of type LoadProhibited occurred on core   0. Exception was unhandled.
Register dump:
PC      :  40081a3e  PS      :  00060031  A0      :  80081784  A1      :  3ffd5700  
A2      :  3ffb1da0  A3      :  0000000d  A4      :  3ffd57b8  A5      :  00400461  
A6      :  ffffffff  A7      :  ffffffff  A8      :  80081a3e  A9      :  3ffd56e0  
A10     :  00000000  A11     :  3ffc54a0  A12     :  3ffc54a0  A13     :  b33f0000  
A14     :  b33fffff  A15     :  00000000  SAR     :  00000017  EXCCAUSE:  0000001c  
EXCVADDR:  00000000  LBEG    :  00000000  LEND    :  00000000  LCOUNT  :  00000000  
Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff9010,len:8
load:0x3fff9018,len:812
ho 0 tail 12 room 4
load:0x40078000,len:3692
load:0x4009f000,len:944
entry 0x4009f1f0
��XTAL 40Mct 20 2016 12:36:18
I (880) phy: phy_version: 246, Nov 18 2016, 17:30:07, 0, 0

Staringlizard, 2016-12-04:
I have the same problem.

mohpor, 2016-12-14:
I'm having random crashes as well.
I have set different settings and found interesting things.
My setup is: 

> (sysname='WiPy', nodename='WiPy', release='0.9.7.b1', version='v1.8.6-234-g765f2bc on 2016-12-02', machine='WiPy with ESP32)

I'm reading a button on `G13`, writing to an active 3V3 buzzer on `G14`
I have set an IRQ on `IRQ_FALLING` for the button and in the handler I'm doing nothing but updating a boolean.

In my main loop (Arduino users beware: you have to make a loop in main by `while True:` syntax  in order to get the loop feature here) I first check if the interrupt has been called, if so, check if it has passed a certain amount of time since last interrupt (a basic debounce, don't know any better solution for it) and if so, buzz!
now, I have a consistent IRQ on my hand, the problem here is, if the button gets pressed and therefore the buzzing event happens too fast or too close to each other, ESP32 will crash complaining about `LoadProhibited`. Now:
* If I press the button far enough from each other, there will be no crashes.
* If I increase my buzzer's pulse width from (currently) 20ms (especially higher than 100ms) it will crash more often.

My assessment here is that ESP32 crashes because it cannot handle the power draw of this particular application and I have probably two options: Limit my power draw by adding a resistor or form a pull down scenarios on `-` pin of the buzzer and power it directly from my power source.

In any case, I would like to know if these crashes are related to power draw or something else. I know ESP32's IRQ handlers doesn't spew out exceptions, but I highly doubt there is an exception in `dowz = True`!

Any ideas?

P.S.: Here is my code:

```python

import machine
import time
from machine import Pin
import micropython
import pycom
import time
import _thread
pycom.heartbeat(False)

from machine import UART
import os
uart = UART(0, 115200)
os.dupterm(uart)

micropython.alloc_emergency_exception_buf(100)
pin = Pin('G13',  mode= Pin.IN,  pull = Pin.PULL_UP)

dowz = False
pval = 0
def cb(p):
    global dowz
    dowz = True
    
pin.irq(trigger = Pin.IRQ_FALLING,  handler = cb)

buzzer = Pin('G14',  mode = Pin.OUT, value = 1,  pull = Pin.PULL_DOWN)

print("Starting the loop")
lastTime = time.ticks_ms()
while True:
    if dowz:
        dowz = False
        now = time.ticks_ms()
        if (lastTime + 300) < now:
            print("I'm Pressed!")
            buzzer(0)
            time.sleep_ms(20)
            buzzer(1)
            lastTime = time.ticks_ms()

```

Staringlizard, 2016-12-16:
I cannot for my life understand how you can even get one interrupt to work. On my board it **crash all the time** when an interrupt is called. I cannot think of a more basic use case than this, and after 16 days nobody even comment about it. Interrupts are not important feature ? lol.

This is my code in **boot.py**:

```
from machine import Pin

import machine
import os
import micropython
import time

button_pressed = False

def button_press(p):
	pass

button = Pin('G17', mode=Pin.IN, pull=Pin.PULL_UP)
button.irq(trigger=Pin.IRQ_RISING, handler=button_press)
```

**I press the button on the expansion board and:**

> Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x40081b02  PS      : 0x00060031  A0      : 0x80081938  A1      : 0x3ffe0b10
> A2      : 0x3ffc1530  A3      : 0x0000000d  A4      : 0x00042445  A5      : 0x00000000
> A6      : 0x00000000  A7      : 0xffffffff  A8      : 0x80081b02  A9      : 0x3ffe0af0
> A10     : 0x00000000  A11     : 0x3ffce660  A12     : 0x3ffce660  A13     : 0xb33f0000
> A14     : 0xb33fffff  A15     : 0x00060f23  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
> 
> Backtrace: 0x40081b02:0x3ffe0b10 0x40081938:0x3ffe0b90 0x40081160:0x3ffe0bc0
> 
>                                                                             Rebooting...

I am now running with Firmware 1.0.0.b1.

danicampora, 2016-12-17:
Hello everyone,

Sorry for the delay on responding this, the reason is that we are working on a whole new interrupt handling mechanism but it is taking a lot more time than we planned.

Cheers,
Daniel

mohpor, 2016-12-17:
@danicampora thanks for the reply.
I hope it gets ready sooner than later. Is it by any chance possible to wake the board from deep sleep using pin interrupts? I'm not even sure if deep sleep is at working stage right now, but theoretically, is it possible? It is very important for us to be able to wake the device from deep sleep using pin interrupts.

danicampora, 2016-12-17:
Yes it is possible. And support for this was added to the IDF like 2 days ago. I am not able to tell you know a firm date for the new interrupt system to be ready, but I hope to be able to do that next week.

Everything you need for your product is possible, it's just a matter of getting the software ready on our side in order to support it.

Cheers,
Daniel

Staringlizard, 2016-12-19:
Thanks for thanks for shedding some light on this.
I am really looking forward to a working version where I can use interrupts. I am dead in the water atm, and I guess many others are too.

Staringlizard, 2017-01-05:
I am now using 1.2.0.b1 and I do not get the crash anymore. However I do not seem to get the handler to execute. This is my code:

```
from machine import Pin

import machine
import os
import micropython
import time

def pin_handler(arg):
    print("got an interrupt in pin %s" % (arg.id()))

p_in = Pin('G17', mode=Pin.IN, pull=Pin.PULL_UP)
p_in.callback(handler=pin_handler, trigger=Pin.IRQ_FALLING | Pin.IRQ_RISING)
```

danicampora, 2017-01-05:
@Staringlizard I'm testing right now on a LoPy-868MHz and it works on both the UART and the Telnet REPL. Are you testing on a WiPy?

danicampora, 2017-01-05:
@Staringlizard I have tested the 3 versions of the firmware (WiPy, LoPy-868 and LoPy-915). All three work fine regarding Pin interrupts. I get the rising and falling edge one.

Can you provide more details about your setup? Do you have anything else connected to ``G17`` ?

Staringlizard, 2017-01-06:
@danicampora I am using Wipy2 with extension board. I do not have anything connected to the any GPIO. I just press the button on the expansion board and I expect to get an interrupt with the code shown. The pin should be G17 on expansion board or P10 on the Wipy2 if I understand it correctly.

danicampora, 2017-01-10:
Already solved with the new interrupt mechanism.

closed 2017-01-10 by danicampora