pycom/pycom-micropython #48: time.time() returns strange value

marisabele, 2016-12-13:
According to documentation time.time() should return seconds since Epoch and running from RTC.
`import time
def timeTest():
     print(time.time())
     time.sleep(60)
     print(time.time())
timeTest()
`
Test code should output timestamps with 60 second difference. In reality diff between timestamps are 377!!!

nevercast, 2017-01-05:
I'm using 1.1.0.b1 on my WiPy 2 and I've found time.time() to return seconds * 10 (including 10th of a second).

Using the same code as you @marisabele my output is 
```
5495
5904
```

a difference of over 400. (Which would be about 40 seconds assuming the RTC is working correctly)

if I change the code to sleep for 1 second.
```python
import time
def timeTest():
    print(time.time())
    time.sleep(1)
    print(time.time())
```

I get varying results of
```
6404
6411

6450
6457

6467
6473

6487
6494
```

Differences of 7, 7, 6, 7.

Which assuming seconds * 10, equates to 7, 7, 6, 7 milliseconds. 

Since I got 40 seconds when sleeping for 60 seconds, and I get 7, 7, 6, 7 milliseconds when sleeping for a second. We are seeing a sleep of about 67% as long as it should be.

So I investigated if the RTC was ticking too slowly or sleep wasn't sleeping long enough and what I got was that time.sleep seems to be very accurate. Testing against a stopwatch the result was often within 40-60ms (a reasonable delay of my own reaction time)

So I'm happy to say that time.sleep() works fine and it is the value being read out from the RTC.
I used another bit of code to collect data on how far out the RTC was. 
```python
def timediff():
  for i in range(5):
    a = time.time()
    time.sleep(20)
    b = time.time()
    dif = b - a
    off = dif / 200 # 200 should be 20 seconds since I know that it's * 10 already.
    print(off)
```

The results I got were [0.68, 0.685, 0.68, 0.68, 0.68]. RTC records 68% of the time that actually elapsed.
I'm not sure how the RTC is configured or if it has a particular oscillator that isn't configured correctly. 

```c
STATIC mp_obj_t time_time(void) {
   // get date and time
   return mp_obj_new_int(mach_rtc_get_us_since_2000() /*/ 1000 / 1000*/);
}
```
This doesn't give me any hints as to the issue. 

One thing that did interest me though was `machrtc.c`

```c
uint64_t mach_rtc_get_us_since_2000(void) {
    uint64_t rtc_ticks;

    // ESP-SDK system_get_rtc_time() only returns uint32 and therefore
    // overflow about every 7:45h.  Thus, we have to check for
    // overflow and handle it.
    rtc_ticks = system_get_rtc_time();
    if (rtc_ticks < rtc_last_ticks) {
        // adjust delta because of RTC overflow.
        delta += UINT64_MAX;
    }
    rtc_last_ticks = rtc_ticks;

    return (uint64_t)(rtc_ticks / 150000);
};
```

The scale 150000, I'm not sure where that comes from. But undoing it in python returns results that look more promising.

```python
def timetest():
  for i in range(5):
    a = time.time() * 150000
    time.sleep(10)
    b = time.time()
    print(b - a)
```

My results were 
``` 
10200000
10200000
10350000
10200000
10200000
```

Assuming a new scale of 1000000 that reads about 10.2 seconds. I would accept 200ms going from Python -> Hal -> Python a few times. (Though I've no data on how quickly Micropython can go between Python and system calls)

So my best assumption on this issue is that the scaler constant in `esp32/mods/machrtc.c` is wrong, or not correct for the RTC on the board I'm using.

Josh.

nevercast, 2017-01-17:
After building the Firmware myself,  time seems to be returning far less strange values (It's certainly seconds now). 

I'm not sure if this is caused by changes to the RTC or a changed scale somewhere?

I'll let someone from Pycom look in to this any further. They may have fixed it?

abiliojr, 2017-01-18:
Yes, that commit fixes the initial bug, and completes most of the RTC functionality

closed 2017-01-18 by abiliojr