pycom/pycom-micropython #49: Not able to catch KeyboardInterrupt exception

RalphHogenbirk, 2016-12-19:
Trying to catch a KeyboardInterrupt that is thrown by a ctrl-c is not working. Running the following code:

```
import time
while (True):
    try:
        time.sleep(2)
    except KeyboardInterrupt:
        print("received keyboard interrupt")
    except:
        print("Got another interrupt")
```

When doing ctrl-c while this is running, the loop breaks and the output is the following, but none of the two prints is performed.
```
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
KeyboardInterrupt: 
>
MicroPython v1.8.6-254-g51066be on 2016-12-19; WiPy with ESP32
Type "help()" for more information.
```

On the WiPy 1.0, it does works; the 'received keyboard interrupt' print is performed like expected.

Running 1.0.0.b2 on a WiPy 2.0 and Pymakr 1.0.0.b7

danicampora, 2017-01-14:
@dpgeorge is this related to having **MICROPY_PY_THREAD** enabled?

danicampora, 2017-01-14:
Doesn't seem to be cause by having threads enabled as the issue is also present with **MICROPY_PY_THREAD = 0**

I'll investigate it...