pycom/pycom-micropython #54: ESP32 compile Error

ouki-wang, 2016-12-30:
CC ../py/../lib/embed/abort_.c
CC ../py/../lib/utils/printf.c
CC main.c
CC mptask.c
CC serverstask.c
CC pycom_config.c
CC mpthreadport.c
mpthreadport.c: In function 'mp_thread_create':
mpthreadport.c:123:23: error: implicit declaration of function 'xTaskCreateStaticPinnedToCore' [-Werror=implicit-function-declaration]
     TaskHandle_t id = xTaskCreateStaticPinnedToCore(freertos_entry, "Thread", *
                       ^
mpthreadport.c:123:23: error: initialization makes pointer from integer without a cast [-Werror=int-conversion]
mpthreadport.c: In function 'mp_thread_mutex_init':
mpthreadport.c:157:21: error: implicit declaration of function 'xSemaphoreCreateMutexStatic' [-Werror=implicit-function-declaration]
     mutex->handle = xSemaphoreCreateMutexStatic(&mutex->buffer);
                     ^
mpthreadport.c:157:19: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
     mutex->handle = xSemaphoreCreateMutexStatic(&mutex->buffer);
                   ^
cc1: all warnings being treated as errors
make: *** [build/WIPY/release/mpthreadport.o] Error 1

robert-hh, 2016-12-30:
You must update the esp-idf too.

ouki-wang, 2017-01-04:
@robert-hh 
thanks for your relay
i have updated the pycom-esp-idf

Compile successful but link error as follow：

AR build/WIPY/release/application.a
CPP build/WIPY/release/esp32_out.ld
LINK build/WIPY/release/application.elf
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_set_config_process+0x18): undefined reference to `pm_check_sleep'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_set_config_process+0x4b): undefined reference to `pm_check_sleep'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_set_config_process':
(.text.wifi_set_config_process+0x166): undefined reference to `pm_check_sleep'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_set_mode_process+0x14): undefined reference to `pm_check_sleep'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.ieee80211_sta_connect+0x1a): undefined reference to `pm_check_sleep'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.ieee80211_sta_disconnect+0xf): more undefined references to `pm_check_sleep' follow
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x8): undefined reference to `pm_disable_delay_timer'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0xc): undefined reference to `pm_stop'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x14): undefined reference to `pm_enable_delay_timer'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x24): undefined reference to `pm_disable_delay_timer'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x2a): undefined reference to `pm_stop'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x49): undefined reference to `pm_enable_delay_timer'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_leave+0x17): undefined reference to `pm_disable_delay_timer'
/home/ouki/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_associated+0x18): undefined reference to `pm_disable_delay_timer'
collect2: error: ld returned 1 exit status
make: *** [build/WIPY/release/application.elf] Error 1

robert-hh, 2017-01-04:
The last two times I had problems updating the esp-idf. There were a few merge conflicts. So I deleted it and mad a fresh clone. And I forgot if I did that and whether it is required, but it cannot hurt. After coling, inside the pycom-esp-idf directory, issue the commands:
```
git submodule update --init
```
and then rebuild the WiPy again in pycom-mycopython/esp32 with
```
make BOARD=WIPY clean
make BOARD=WIPY
```
I just made a trial build for both WiPy and LoPy, w/o any issues.

ouki-wang, 2017-01-06:
hi @robert-hh
I get the lastest esp-idf, but still failed to connect

ouki@ubuntu:~/esp/pycom-esp-idf$ rm * -rf
ouki@ubuntu:~/esp/pycom-esp-idf$ git checkout .
ouki@ubuntu:~/esp/pycom-esp-idf$ git pull
Already up-to-date.
ouki@ubuntu:~/esp/pycom-esp-idf$ git submodule update --init
Submodule path 'components/bt/lib': checked out '91657e0c4025f5a694b0a89f449c347b0f2fdf79'
Submodule path 'components/esp32/lib': checked out '02232f974b0ff1568ddd6d7015a41fb4f4870994'
Submodule path 'components/esptool_py/esptool': checked out 'adc914b91ac6d2cfd1ace56307b4374eb9439e14'
Submodule path 'components/micro-ecc/micro-ecc': checked out '14222e062d77f45321676e813d9525f32a88e8fa'
ouki@ubuntu:~/esp/pycom-esp-idf$ git remote show origin
* remote origin
  Fetch URL: https://github.com/pycom/pycom-esp-idf
  Push  URL: https://github.com/pycom/pycom-esp-idf
  HEAD branch: master
  Remote branch:
    master tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (up to date)

LINK build/WIPY/release/application.elf
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_scan.o):(.text.scan_start+0x4c): undefined reference to `pm_check_sleep'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x8): undefined reference to `pm_disable_delay_timer'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0xc): undefined reference to `pm_stop'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x14): undefined reference to `pm_enable_delay_timer'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x24): undefined reference to `pm_disable_delay_timer'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x2a): undefined reference to `pm_stop'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_pm$part$6+0x49): undefined reference to `pm_enable_delay_timer'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_leave+0x17): undefined reference to `pm_disable_delay_timer'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_sta_associated+0x18): undefined reference to `pm_disable_delay_timer'
/home/dfrobot/esp/pycom-esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_set_config_process+0x18): undefined reference to `pm_c

ouki-wang, 2017-01-06:
After the update sdk, I compile esp32 successfully. thanks

closed 2017-01-31 by abiliojr