pycom/pycom-micropython #55: wlan.disconnect() doesn't work

GregS2, 2017-01-01:
wlan.disconnect() doesn't seem to work - the interface remains connected.
Serial REPL log:
~~~
Connecting to a Pycom device...
Connected
>>> wlan.isconnected()
True
>>> wlan.disconnect()
I (845584) wifi: state: run -> init (0)
I (845585) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:6
I (845585) wifi: pm stop, total sleep time: 0/84065182

>>> I (845709) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:6
I (845709) wifi: state: init -> auth (b0)
I (845712) wifi: state: auth -> assoc (0)
I (845716) wifi: state: assoc -> run (10)
I (845761) wifi: connected with WLAN, channel 11
I (855716) wifi: pm start, type:0

>>> wlan.isconnected()
True 
~~~
Board: LoPy, f/w 1.1.0.b1

mjmare, 2017-01-04:
Same here with same board en f/w.

nevercast, 2017-01-05:
I can assume that it's due to auto-re-association

https://github.com/pycom/pycom-micropython/blob/master/esp32/mods/modwlan.c#L295-L299

A state variable should probably be used to verify if we want to be associated or not.

danicampora, 2017-01-14:
Fixed here: https://github.com/pycom/pycom-micropython/commit/15453d6b02238e1c27262b8886be2a7b09f13aeb

Thanks to @dpgeorge and @nevercast for the hints.

closed 2017-01-14 by danicampora