pycom/pycom-micropython #59: fail to use micropython on esp32

yuxiangyuzm, 2017-01-12:
when i flash the wipy.bin(it's a file what we can use python on esp32 ) into esp32, it is successful, but when i use it,  there are something wrong, just like this(on putty):  
                               >a=3
                               >print(a)
                               3
                               >import os
                               what?
                               >import sys
                               what?
                               >
i do not know why. 
  [image]

Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
```
The output will look similar to this:
```
(sysname='LoPy', nodename='LoPy', release='0.9.4.b1', version='bc42ea7 on 2016-11-14', machine='LoPy with ESP32')
```
* Exact steps to cause this issue
    1. First I did this...
    2. Then, I did that...
* What you expected
    * e.g. I expected *abc* to *xyz*
* What happened instead
    * e.g. Instead, I got *zyx*

Thank you!

nevercast, 2017-01-12:
I thought the REPL had >>> not >

yuxiangyuzm, 2017-01-12:
yes, but what result in this problem. the bin file?or something esle? @nevercast

nevercast, 2017-01-12:
Try using the latest firmware. Have you used the firmware updater?

yuxiangyuzm, 2017-01-12:
@nevercast ok i  will have a try , but two days ago ,i did this. i think it's latest firmware. thank you!

yuxiangyuzm, 2017-01-12:
@nevercast i tried just now. there is also this problem. 
[image]

negativekelvin, 2017-01-14:
@nevercast yes that's the tinybasic prompt

Its a flash configuration issue. Suggest to build firmware from source for running on non-pycom hardware

DirkHeinke, 2017-03-20:
Any success running pycom-micropython on a ESP32 dev board?