pycom/pycom-micropython #60: Include Errno for OSErrors

PMunch, 2017-01-12:
I'm using the LoPy boards with the latest firmware 1.2.2.b1 and trying to use mpfshell to communicate with the device over serial. The problem mpfshell has when working with the LoPy boards is that when using the os functions such as os.listdir, os.remove, os.open, etc. any error is a simple OSError with only a textual description of the problem. This message however seems to differ between versions of MicroPython and so the textual description is not enough for an automated script to understand which error has actually occured. For example the mpfshell function for rm:
```
try:
    self.eval("os.remove('%s')" % self._fqn(target))
except PyboardError as e:
    if "ENOENT" in str(e):
        raise RemoteIOError("No such file or directory: %s" % target)
    elif "EACCES" in str(e):
        raise RemoteIOError("Directory not empty: %s" % target)
    else:
        raise e
```
This works fine when the Errno is included and the error message contains the string "ENOENT" or "EACCES" but for this version the ENOENT message is replaced by "no such file or directory: " which is not correctly identified.

It would be a great improvement to add an errno field to OSError (or rather use it properly, it already exists but points to the error message instead) so the exceptions are machine readable. This also makes it easy to write exception handlers that responds differently to different errors.

danicampora, 2017-01-12:
@PMunch thanks, this is a very good point. We will modify the exceptions thrown to include the error number.

BTW there's already an **errno** module. Try:

```python
import errno
dir(errno)
```

PMunch, 2017-01-12:
Yeah I saw the errno module, but it's not a whole lot of use when you can't get the actual error number from the exception. Looking forward to better errors.

dpgeorge, 2017-01-13:
> We will modify the exceptions thrown to include the error number.

OSError's should always print the errno.  In the latest uPy upstream version (v1.8.7) OSError will be printed as one of the following:
```
>>> raise OSError(2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```
or
```
>>> raise OSError(20)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 20
```
Ie the errno is always printed, either in the form `OSError: [Errno <int>]` or `OSError: <int>`.  If the errno is unknown, or if the build doesn't include the uerrno module, then the latter form is used, else the former.  If your code matches both of these then it should be robust and work for all uPy ports.

PMunch, 2017-01-13:
Problem is that in this version of µPy OSError is raised with a string [like so](https://github.com/pycom/pycom-micropython/blob/dac16bdf774d29528ffb1f099a56879a71d7d124/esp32/mods/moduos.c#L235). This means that the error message becomes "OSError: no such file or directory". Looking at the exceptions .errno value it is set to the same string. This makes it really inconvenient to handle these exceptions.