pycom/pycom-micropython #71: utime.time Not Returning Large Numbers Properly

dchappel2, 2017-01-26:
sysname='LoPy', nodename='LoPy', release='1.3.0.b1', version='v1.8.6-379-gc44ebac on 2017-01-13', machine='LoPy with ESP32')

Steps Taken:

1. Initialized RTC with current time.

2. rtc.now() value is correct 
   ex. (2017, 1, 26, 18, 18, 46, 828743, None) 
   and updates properly.  EXPECTED

3. utime.time() value is frozen at the initialization time ex. 1485454720 and does not update.  NOT EXPECTED

Alternative attempt:

1. Reboot the unit and DO NOT initialize the RTC (i.e. 1/1/1970)

2. Both rtc.now and utime.time() update as expected with the small base number.

Additional backup info in forum at: forum.pycom.io/topic/559/unix-time

This is needed to enable sending full 10 digit Unix timestamp in a string.

jmarcelino, 2017-01-27:
The problem is in esp32/mods/modutime.c
```
/// \function time()
/// Returns the number of seconds, as an integer, since 1/1/1970.
STATIC mp_obj_t time_time(void) {
   // get date and time
   return mp_obj_new_float(mach_rtc_get_us_since_epoch() / 1000000.0);
}
MP_DEFINE_CONST_FUN_OBJ_0(time_time_obj, time_time);
```

I don't understand why is this returning a float? Should be an int as the description suggests right?

By simply changing it to return an int (and use integer division)

```
return mp_obj_new_int(mach_rtc_get_us_since_epoch() / 1000000);
```

 it began working correctly for me:

```
>>> rtc.init((2017,1,27,9,0))

Running utimeTest.py
rtc.now:(2017, 1, 27, 9, 0, 9, 37484, None)  utime.time:1485507609
rtc.now:(2017, 1, 27, 9, 0, 10, 39594, None)  utime.time:1485507610
rtc.now:(2017, 1, 27, 9, 0, 11, 41610, None)  utime.time:1485507611
rtc.now:(2017, 1, 27, 9, 0, 12, 43605, None)  utime.time:1485507612
rtc.now:(2017, 1, 27, 9, 0, 13, 45611, None)  utime.time:1485507613
rtc.now:(2017, 1, 27, 9, 0, 14, 48617, None)  utime.time:1485507614
rtc.now:(2017, 1, 27, 9, 0, 15, 50614, None)  utime.time:1485507615
rtc.now:(2017, 1, 27, 9, 0, 16, 52619, None)  utime.time:1485507616
```

Am I missing some important detail

abiliojr, 2017-01-27:
The problem seems to be Espressif's tools. There seems to be an issue when passing floats bigger than 2^24 as a parameter to a function. See more [here](https://forum.pycom.io/topic/559/unix-time/10)

dpgeorge, 2017-01-27:
Single precision floats don't have enough bits to represent the number of seconds since the Epoch.

abiliojr, 2017-01-28:
Yes. For now on the result will be an int as other implementations of micropython do. This could be changed back into a `double` in the future.

closed 2017-01-31 by abiliojr