pycom/pycom-micropython #80: FTP Server support of files/folders names containing spaces or accents

cvpe, 2017-02-16:
Wipy2, all versions

Is it possible to modify the FTP server to support file or folder names containing spaces or accents. The OS supports this kind of files/folders because I can create them in REPL, thus it should be possible because any server program uses the same instructions to create a file or a folder.

cvpe, 2017-02-16:
For me it's a bug because you're able to create theses folders and files but the FTP Server does not allow to reach them... But anyway, I'ld be happy to get an update 😁

Bucknalla, 2017-02-16:
Good point!

cvpe, 2017-02-22:
Do you know when I could hope to see it working?

Bucknalla, 2017-02-23:
Hi @cvpe, @danicampora and the team will likely have logged this bug now and it will have made it onto the roadmap - currently our priorities are enabling deep sleep mode on the ESP32. I'll get back to you when I know more!

cvpe, 2017-05-17:
@Bucknalla still no news 3 months later 😞