pycom/pycom-micropython #81: Support of SD card bigger than 32GB

cvpe, 2017-02-16:
Wipy2, all versions

Could support SD card of more than 32GB
I had asked that via a mail to Pycom support and the answer had be yes, some months ago, before I buy the Wipy2.

danicampora, 2017-02-22:
This will come for free once we merge our code with the official MicroPython repo as they have exFAT enabled there which supports SD cards larger than 32 GB.

dpgeorge, 2017-02-22:
Note that exFAT is supported but not enabled by default.  There are licensing issues (with Microsoft, who hold patents on exFAT) that means you can probably only use exFAT if you have obtained a license.

danicampora, 2017-02-22:
@dpgeorge good point!

cvpe, 2017-02-22:
And then?

cvpe, 2017-03-20:
I really don't understand because you can have FAT32 with SD cards of more than 32 GB, like 64GB.
FAT32 supports until 2TB