pycom/pycom-micropython #82: Lopy hangs completely with gc.collect()

JF002, 2017-02-18:
I'm using a Lopy board with firmware version 1.6.3.b2 : 
```
>>> import os; [os.uname().sysname, os.uname().machine, os.uname().release]
['LoPy', 'LoPy with ESP32', '1.6.3.b2']
```
For my application, I read temperature from a DS18B20 and send it to a MQTT broker via Wifi.
The system hangs after a few minutes. I noticed that the free RAM dropped quickly, so I tried to add gc.collect() into my main loop, but it seem to hang the system.

Here is a little code sample that reproduce the issue, with no wifi, mqtt, onewire,...:

```
import gc
import time

print("Starting the loop")

while True: 
    print("loop...")
    time.sleep(1)
    gc.collect()
    
print("End of the loop")
```
After the first call to gc.collec(), the REPL does not respond anymore, the script that was running shows no sign of activity, and the heartbeat LED does not blink anymore.
Nothing is printed on the console.

robert-hh, 2017-02-18:
With this little test program of JF002, loaded to the LoPy as test.py and executed via "import test",  I see the following:
- after hard reset (push reset button) it always runs
- after warm reset (Ctrl-D) it always hangs.

JF002, 2017-02-27:
I've run my script with firmware version 1.6.5.b1, and gc.collect did not crash as it did before.
I'll see if it can run for a longer time, but it seems that this bug is now fixed.