pycom/pycom-micropython #84: Since firmware upgrade, UART.readline() now reads less than full lines.

Pete-v, 2017-02-25:
`(sysname='WiPy', nodename='WiPy', release='1.6.4.b1', version='v1.8.6-475-g2320cb2d on 2017-02-24', machine='WiPy with ESP32')`

Since upgrading from 1.5.1, I've noticed a change (regression?) in the UART API.

I'm not sure if it's strictly correct, but with the earlier firmware I was doing:
```
def readUart():
    while True:
        if uart.any():
            line = str(uart.readline(), "utf-8")
            # do stuff with line
```
and reliably getting a newline-terminated chunk of input from the serial line each time.

With the new firmware, the same code (with the same serial input) often populates `line` with a fragment of the actual line, breaking at a random point where there certainly isn't a newline. The next read will continue with the rest of the line. They're breaking into random chunks of roughly 3 to 25 characters. I haven't noticed any instances of it failing to break on a genuine newline. I'm reproducing this with a minimal case where "do stuff" is just printing `line` in the REPL.

Skipping the `uart.any()` test makes no significant difference, `readline` just returns streams of `None` while there's no data incoming. When data does arrive, it's broken at random imaginary linebreaks as above.

danicampora, 2017-02-25:
@Pete-v thanks, will investigate now. It's weird since we haven't touched anything around the UART drivers between those 2 releases.

Pete-v, 2017-02-25:
Thanks. Maybe it's some kind of timing thing - like maybe the line-splitting never worked properly, but the old firmware was slower and so the buffer was more likely to be full when read?

danicampora, 2017-02-25:
@Pete-v I have tested it and the behaviour is correct. We did change something between those releases and was the timeout value for the UART Rx. Before it was 16 characters, which was extremely high, and we reduced it to 2 characters, and this is what is causing the behaviour you see. However, if I do your test and I paste a single full line on the REPL, I always get the complete line. What you see is probably caused because the way you feed the characters insert a delay of more than 2 characters. I think we should add en extra parameter to the UART constructor in order to be able to specify the number of characters to wait before a UART Rx timeout occurs. What do you think?

danicampora, 2017-02-25:
BTW, I have updated the docs of the readline() method which was incomplete before:

https://docs.pycom.io/pycom_esp32/library/machine.UART.html?highlight=readline#machine.uart.readline

Pete-v, 2017-02-25:
Thanks for looking into this so quickly.

Just checking I understand this right - you're saying that sometimes the device I'm receiving from waits slightly too long before sending the next character, and this causes readline to return early?

Unfortunately I have no control over the sending device, so if this is the problem then it does sound like the configurable timeout is required. I note that the [pyboard version of UART](https://docs.micropython.org/en/latest/pyboard/library/pyb.UART.html#pyb.UART.init) does have this.

Thanks,

Pete

Pete-v, 2017-03-01:
I see the timeout argument is in the new firmware - haven't had time to try it yet, but thanks for the fast turnaround!