pycom/pycom-micropython #87: Your Timer tutorial does not work with pymakr

JosephAFuentes, 2017-03-07:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
```
The output will look similar to this:
```
(sysname='LoPy', nodename='LoPy', release='0.9.4.b1', version='bc42ea7 on 2016-11-14', machine='LoPy with ESP32')
```
* Exact steps to cause this issue
    1. First I did this...
    2. Then, I did that...
* What you expected
    * e.g. I expected *abc* to *xyz*
* What happened instead
    * e.g. Instead, I got *zyx*

Thank you!

Bucknalla, 2017-03-07:
Hi @JosephAFuentes could you please post a snippet of your code along with specifically what was failing?

Thanks!

Xykon, 2017-12-19:
Closing as no further feedback received.

closed 2017-12-19 by Xykon