pycom/pycom-micropython #89: Micropython for LoPy compilation fail on Ubuntu 15.04 (32 bits)

karolalexandersecret, 2017-03-26:
Hi,

When compiling Micropython for LoPy on a 32 bits Ubuntu 15.04 Laptop. Start Compilation with:
`make BOARD=LOPY LORA_BAND=USE_BAND_868 flash`
 fail with a lot of errors like this:
`In file included from ./mpconfigport.h:90:0,
                 from ../py/mpconfig.h:45,
                 from ../py/mpstate.h:31,
                 from ../py/mpstate.c:27:
/home/Alex/esp32workplace/pycom-esp-idf/components/freertos/include/freertos/FreeRTOS.h:244:3: error: #error configMAX_TASK_NAME_LEN must be set to a minimum of 1 in FreeRTOSConfig.h
  #error configMAX_TASK_NAME_LEN must be set to a minimum of 1 in FreeRTOSConfig.h`

For my other Laptop running 64 bits Ubuntu, everythink work well.

Tried:
-Compiling programs from the example/get-started folder ===> Work well !
-Reinstaling Compiler, pycom-esp-idf and pycom-micropython folder ===> Nothink change
-Delete in source the part that generate error ===> Create another errors
-Trying BOARD=WIPY ===> Fail

Cannot find where does the problem came from. Can someone help me ?

Alexander

danicampora, 2017-03-26:
Are you building against this IDF: https://github.com/pycom/pycom-esp-idf ?

karolalexandersecret, 2017-03-26:
Hi Daniel,

Yes, I also tried the original one, but without success :/

jmarcelino, 2017-03-26:
Shouldn't you be compiling https://github.com/pycom/pycom-micropython-sigfox instead ?

I'm not sure this older repo matches the latest pycom-esp-idf

karolalexandersecret, 2017-03-26:
Hi Jose,

You right, now better. But terminate with error:
`python ../py/makeversionhdr.py build/LOPY/release/genhdr/mpversion.h
MPY frozen/_boot.py
mkdir -p build/LOPY/release/frozen_mpy/
../mpy-cross/mpy-cross -o build/LOPY/release/frozen_mpy/_boot.mpy -s _boot.py  frozen/_boot.py
bash: ../mpy-cross/mpy-cross: Aucun fichier ou dossier de ce type
../py/mkrules.mk:116: recipe for target 'build/LOPY/release/frozen_mpy/_boot.mpy' failed
make: *** [build/LOPY/release/frozen_mpy/_boot.mpy] Error 127`

jmarcelino, 2017-03-26:
Make sure you're following the new build instructions at https://github.com/pycom/pycom-micropython-sigfox/blob/master/esp32/README.md

There's an extra step now:

First build the mpy-cross compiler:

$ cd ../mpy-cross
$ make all

karolalexandersecret, 2017-03-26:
My bad, not seen them
Now close to the end. Work well until the final step:
`...
...
CC util/mpirq.c
CC util/mpsleep.c
CC util/timeutils.c
CC telnet/telnet.c
CC ftp/ftp.c
CC ftp/updater.c
CC build/LOPY/release/pins.c
AR build/LOPY/release/application.a
CPP build/LOPY/release/esp32_out.ld
LINK build/LOPY/release/application.elf
   text	   data	    bss	    dec	    hex	filename
 966781	 169828	  73048	1209657	 127539	build/LOPY/release/application.elf
Building partitions from lib/partitions.csv...
IMAGE build/LOPY/release/lopy_868.bin
esptool.py v2.0-beta2
Signing OTA image
make: *** No rule to make target 'build/LOPY/release/bootloader/bootloader.bin', needed by 'flash'. Arrêt.
`

karolalexandersecret, 2017-03-26:
Okay problem came from the second instruction that clean boot creation process:
```
make BOARD=LOPY -j5 TARGET=boot
make BOARD=LOPY -j5 TARGET=app clean <=== don't use
```

Now work well.
Thank you very much Jose

closed 2017-03-27 by Bucknalla