pycom/pycom-micropython #91: Failing to compile the ESP32 version

crazyquark, 2017-05-07:
I am trying to compile the ESP32 version for a WiPy 2.0 on OS X Sierra.

I set everything up as described in the readme but there seems to be a configuration error, I get this compiler error:
[...] pycom-esp-idf/components/freertos/include/freertos/FreeRTOS.h:244:3: error: #error configMAX_TASK_NAME_LEN must be set to a minimum of 1 in FreeRTOSConfig.h
  #error configMAX_TASK_NAME_LEN must be set to a minimum of 1 in FreeRTOSConfig.h

danicampora, 2017-05-07:
Don't use this repo. Use pycom-micropython-sigfox instead.

jmarcelino, 2017-05-07:
@danicampora 

Can you add a big DO NOT USE - THIS IN AN OLD VERSION 
Use pycom-micropython-sigfox instead .

to the top level readme? :)

Lots of people seems confused by the pycom-micropython-sigfox, they think it's only for Sigfox.

danicampora, 2017-05-07:
@jmarcelino done, thanks :-)

nevercast, 2017-05-24:
This is still an issue on the latest sigfox build. configMAX_TASK_NAME_LEN becomes literally `CONFIG_FREERTOS_MAX_TASK_NAME_LEN` which triggers FreeRTOS.h error for `< 1` because a string is technically `0` when compared like this.

This seems to be because `sdkconfig.h` isn't "working". I'm currently trying to fix this but not having a great time. I surrounded ` #define configMAX_TASK_NAME_LEN			( CONFIG_FREERTOS_MAX_TASK_NAME_LEN )` with `#ifdef` like this:

```
#ifdef CONFIG_FREERTOS_MAX_TASK_NAME_LEN 
#define configMAX_TASK_NAME_LEN			( CONFIG_FREERTOS_MAX_TASK_NAME_LEN )
#endif
```

Which got the compiler to carry on as `configMAX_TASK_NAME_LEN` defaults to 16 if undefined. However it just broke later on when the compiler required some `CONFIG_WLAN_*` macros for esp_wifi.

I'll create an issue from pycom/pycom-micropython-sigfox pointing to this one in case anyone else is experiencing this issue.

Xykon, 2017-12-19:
Closing this issue.

closed 2017-12-19 by Xykon