pycom/pycom-micropython #92: Can't sync files

keenedge, 2017-05-22:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
================
LoPy

* The firmware version you are using.                                                          
==================
( sysname='LoPy', nodename='LoPy', release='1.6.13.b1', version='v1.8.6-607-g9c8a0e9e on 2017-05-01', machine='LoPy with ESP32', lorawan='1.0.0')                                                                                                 

Problem
============
run this code
print("main")

print("main")

from network import Bluetooth
import time

bt = Bluetooth()

bt.init()

if( not bt.isscanning()):
    bt.start_scan(-1)

time.sleep(1)
adv = bt.get_adv()

if( adv ):
    print("1")
    print(adv.mac)
    print("2")
#    bt.stop_scan()
    print("3")
    conn = bt.connect(adv.mac)
    print("4")
    if conn.isconnected():
        print("5")
        print("Connected")
        print("6")
    else:
        print("7")
        print("Not Connected")
        print("8")

    s = conn.services()
    s
print("9")

bt.stop_scan()
print("10")
* What you expected
  "Connected""

* What happened instead

The output is this:                                                                                                                   
1                                                                                                                       
b'\xef2M\xc0]\xe1'                                                                                                      
2                                                                                                                       
3  

Program hangs on bt.connect()

armageddon421, 2017-07-18:
I'm having the exact same problem with the latest firmware. I have only one BLE peripheral device to test it with, but that works fine with my smartphone. Please let me know if you found out what causes this.

armageddon421, 2017-07-18:
removing "bt.init()" seems to have solved the problem for me. Strange.